sandeepmistry/node-bbc-microbit #2: Must work with paired micro:bit

bluetooth-mdw, 2016-07-06:
By default, micro:bits require pairing. Currently the only way to disable this security requirement is to use mbed or yotta as opposed to the more popular on-line editors like PXT. I think it's important that this (marvellous) API be usable wit any micro:bit regardless of how its firmware was created.

I paired my micro:bit with BlueZ's bluetoothctl, naively assuming that my node.js script would then magically work. But testing against the paired micro:bit I still see that running my node.js script is triggering an attempt to pair (which fails since the micro:bit is not in pairing mode).

sandeepmistry, 2016-07-18:
Hi @bluetooth-mdw,

This is because noble uses it's own built-in SMP paring stack, and is not aware of anything stored in the kernel. noble would have to be changed to use the kernel's SMP stack instead, which is not something that will likely be done in the short term.

For now, I thinks it's ok for users to flash a custom hex file that does not require pairing. There's one included in this repo.

I think there also needs to be a bigger discussion about the micro:bit and pairing. Right now the model works well with Android and Linux, as the system apps support pairing with any BLE device that is advertising. However, both the iOS and OS X system settings apps will only list classic devices or BLE devices advertising special 16-bit UUID's like HID over GATT in the Bluetooth settings pairing screen.

So for someone to pair with iOS or OS X now, they need to put the micro:bit into pairing mode, then get generic BLE app like LightBlue to try to read a secured characteristic on the micro:bit. This will cause a pin dialog to appear. I think this flow is quite awkward. Alternatively, with OS X and noble another script can be created for the paring process.

Any thoughts on this?

bluetooth-mdw, 2016-07-18:
In the UK, pairing requirements for micro:bit effectively came from the BBC with classroom and some child protection scenarios in mind. So I doubt that microbit.co.uk will change the way it generates code with respect to Bluetooth pairing. Elsewhere (e.g. Microsoft's PXT) I am actively reviewing this and hoping to move to either Just Works pairing or No Pairing as the default but provide a code Block with which pairing requirements can be set by the user. Either way, some micro:bits will require pairing and some will not.

Agreed, It will sometimes be OK to just substitute a hex file that does not require pairing, created using Yotta or mbed. But sometimes it will not. It depends whether you're using node.js for something serious and which has security requirements or if you're just playing I guess.

sandeepmistry, 2016-07-24:
I totally understand the need for pairing security in a classroom scenario.

However, we should also keep OS X (as well as iOS) users in mind, just not Linux/Raspberry Pi when looking into adding support for pairing.

bluetooth-mdw, 2016-07-25:
Totally agree. After all, ultimately this is really about noble in general rather than micro:bit specifically. Support for security in noble would make it a candidate for much more serious IoT implementations.

sandeepmistry, 2016-07-27:
Getting the ball rolling on recommended pairing procedure for OS X: https://github.com/lancaster-university/microbit-dal/issues/187

One option is to have a Node.js app that initiates pairing on both Linux and OS X. Then the user can enter the pin in the terminal window. Need more research to see if this is possible on OS X. However, it would still be great if the OS X settings app could handle the pairing for us.

bluetooth-mdw, 2016-07-27:
Sounds good. We have pairing working via a web UI on Raspberry Pi with our Bluetooth Internet Gateway. I'll send you a link by email just fyi.

Doesn't BlueZ use the concept of agents for things like this?

hansamann, 2017-01-29:
Hi @sandeepmistry and @bluetooth-mdw - I just explored the microbit & BLE and basically run into the issues you both luckily discussed publicly here. For a quick moment, I was somehow able to use the button service, without special need for the .hex file in the firmware folder. I played with Lightblue and the microbit was in pairing mode, I sent some val I think to soome characteristic and boom it showed the OK sign and later I was able to use the button-listener example js file to get these events ( I would also see the characteristic change in LightBlue). 

But now I am failing to do this trick again :-) and if course it is not  very user-friendly or reliable to somehow trick the microbit. So I am wondering if there is some news on this front. I also read in this thread about raspberry PI etc - is it working there due to the way the Raspberry PI handles pairing? 

I ultimately want to put my own logic on the bbc microbit and enable the BLE services that I need. For example I would want to count the beats of a heart rate sensor that I connected via wires and from time to time distribute the bpm number via the uart service. For this, I need more than a customer .hex file (that I cannot import into the editor for adding the custom part). 

thx for the great work so far and thx for getting back to me with an update!

bluetooth-mdw, 2017-01-29:
I don't know what you had on your micro:bit but what you described sounds like "Just Works" pairing, which is an approach to pairing now supported in the micro:bit firmware. And indeed, writing to a characteristic which has security requirements, would trigger the pairing process.

Typically I pair my micro:bit with Raspberry Pi using BlueZ commands.

As for putting your own logic on the micro:bit, see https://pxt.microbit.org/ and https://lancaster-university.github.io/microbit-docs/

deadprogram, 2017-02-07:
Is there somewhere where the source code for the hex file in this repo is located?

mcgreal, 2017-02-11:
Hi all,

I've been doing a project using the MicroBit whereby I need to communicate with my Windows 10 laptop via Bluetooth.

I can connect with the MicroBit having installed noble and following the steps to configure the BLE Adapter and I get the following message when I run the button-listener.js example:
connecting to microbit
              connected to microbit
subscribing to buttons
              subscribed to buttons

But my MicroBit remains in pairing mode meaning I cannot press the buttons (well, I can physcially press them but they don't send acknowledgements to the program).

I know with Linux that it is possible to pair using BlueZ whereby the 6 digit passcode can be entered.
My question is, can I do the equivalent on Windows?

Before configuring the BLE Adapter, I was able to see the MicroBit listed in my Device Manager as a Bluetooth device. From there I could click pair and enter the 6 digit pin but this is not possible since configuring the adapter. In essence I am "connected" but not "paired".

Any suggestions?

Thanks,
Ciarán

AlexKoe, 2017-02-14:
Hi all,

first of all thanks a lot to @sandeepmistry for creating that cool bbc microbit node.  ;)
The node works perfectly as long as I use the custom .hex file as provided in the firmware folder.

As soon as I am developing a custom bluetooth application using https://pxt.microbit.org,  I am having problems connecting through node-bbc-microbit from my Raspberry Pi2 to the microbit.

Luckily https://pxt.microbit.org now allows to set in the project's pxt.json file the option 'Unsecure: Anyone can connect via Bluetooth' to allow connections w/o a complex pairing step.

bluetoothctl on the RPi2 is able to 'see' the microbit and is able to connect to the microbit w/o any major problems. The same is true for the micro:bit Blue app on my Android smart phone.

As soon as I start my node-bbc-microbit based .js script, it doesn't seem to find the microbit (and yes, I did disconnect bluetoothctl before) and hence I get no connection.

Any ideas?

@sandeepmistry : would you be so kind and share the source code of your .hex file with us please?  ;)

Thanks in advance and greetings, Alexander

pelikhan, 2017-02-14:
@sandeepmistry you can create a script on pxt.microbit.org and use the "share project" to create a deep link to the source (gearwheel menu -> share).

deadprogram, 2017-02-18:
For anyone who is looking seems like this project is using the default Bluetooth LE profile in the standard firmware: https://github.com/lancaster-university/microbit-dal

Hope that helps!

sandeepmistry, 2017-03-13:
Firmware source uploaded to Github: https://github.com/sandeepmistry/node-bbc-microbit-firmware