sandeepmistry/node-bbc-microbit #6: Source code of the firmware

vshymanskyy, 2016-10-23:
Could you also share the source code of the firmware?

bluetooth-mdw, 2016-10-23:
You can use either C/C++ or Microsoft's PXT tool at https://pxt.microbit.org/

You'll find information on this topic at:

http://bluetooth-mdw.blogspot.co.uk/p/bbc-microbit.html 

http://www.bittysoftware.com

and there's an example project in the micro:bit sample repo here:

https://github.com/lancaster-university/microbit-samples/tree/master/source/examples/bluetooth-services 

Cheers

Martin

sandeepmistry, 2016-10-23:
@bluetooth-mdw thanks for helping out!

@vshymanskyy I've attached the mbed exported code from the online version of mbed, it's based on the micro:bit template.

[node-bbc-microbit_zip_nrf51_microbit.zip](https://github.com/sandeepmistry/node-bbc-microbit/files/546610/node-bbc-microbit_zip_nrf51_microbit.zip)

closed 2016-10-23 by sandeepmistry