sandeepmistry/node-bbc-microbit #10: micro:bit with chip (next thing co)

tomasmcm, 2016-12-26:
I've been testing this amazing library with a mac and a pocket chip. My experience with the mac has been butter smooth, however with the chip it has been complicated...

I've tried multiple tutorials, this one being the best put together https://www.hackster.io/inmyorbit/connect-a-mobile-application-to-your-c-h-i-p-using-ble-96fe96, but nothing worked, there were no errors, no feedback at all, it just hang...

Then, while testing some random commands from here https://github.com/sandeepmistry/noble#running-on-linux I discovered that if I added `NOBLE_REPORT_ALL_HCI_EVENT1` between `sudo` and `node` it started working!

`sudo NOBLE_REPORT_ALL_HCI_EVENT1 node index.js`

So, I'm not sure if this is a bug or a feature, but I've seen other people with a similar problem so I though I would share this, and maybe adding a note on the README would be nice too.

sandeepmistry, 2017-01-03:
@tomasmcm this should be ok when a new version of noble is published with the following change: https://github.com/sandeepmistry/noble/pull/477

Could you please try manually updating noble to the master version? `npm install sandeepmistry/noble`

tomasmcm, 2017-02-04:
@sandeepmistry Thanks!
Manually updating to the master version worked, my C.H.I.P can now detect and pair with the micro:bit.

closed 2017-02-04 by tomasmcm