sandeepmistry/node-bbc-microbit #11: Firmware issue?

bjablonskipwr, 2017-01-25:
Great library - thanks for creating this!

I'm trying out examples for the start. I've started with accelerometer-service.js for the start. It seems that microbit get disconnected as soon as the "Draw a circle" procedure finishes on the device. Is it some firmware issue or bluetooth stack problem?

bluetooth-mdw, 2017-01-26:
It's not a Bluetooth stack problem.

bluetooth-mdw, 2017-01-27:
I think this is a bug in the micro:bit's "DAL" i.e. Device Abstraction Layer. I've discussed with Lancaster University, who own the DAL and submitted an issue:

https://github.com/lancaster-university/microbit-dal/issues/262

sandeepmistry, 2017-01-30:
@bjablonskipwr I think I had similar issues.

I had to calibrate the magnetometer right after flashing the firmware or things would be flaky connection wise. This is noted in the read me.

Closing this for now, as the issue is reported upstream.

closed 2017-01-30 by sandeepmistry