sandeepmistry/node-bbc-microbit #12: microbit burn through battery

curioussavage, 2017-01-27:
Hi!

I'm using the microbit with noble - not using this code specifically. I am using the firmware provided here. I'm using a cr2032 coin cell to power the chip. But it has burned through the battery in a single day twice now! I'm not sure how to troubleshoot this. Do you have an idea?

I'm doing the following: 

```

buttonCharacteristic.subscribe(callback);

buttonCharacteristic.on('data', function(data) {
  // do stuff
})

```

other than listening to `disconnect` for peripheral that's all.

Between presses the microbit shouldn't be sending any data and thus not be using much power, or at least so I thought.

curioussavage, 2017-01-27:
not really related but I'm hoping to use [these](http://www.banggood.com/Pet-Dog-Anti-lost-Tracker-Smart-Buletooth-Tracer-GPS-Locator-Tag-Alarm-Tracer-Finder-p-1078311.html?rmmds=detail-top-buytogether) instead in the future.

finneyj, 2017-01-31:
@curioussavage That's really unexpected - I've found micro:bit to be generally very power efficient... 

Thanks for flagging this up - we should check out that firmware build in case there's something odd in there. Was the firmware driving the LEDs on the board or anything?

finneyj, 2017-01-31:
@sandeepmistry @curioussavage 

For info - a snapshot of the last power analysis on the micro:bit that I've seen below... the sensors are very low power (uW range), so it looks like there may be something odd going on here. To burn a CR2032 in a day, we'd have to be pulling around 30mW, which is way up there compared to it's measured idle current.

Have either of you got the source code for the firmware you're using? If so, I can take a look....

[image]

Do either of you have the source

bluetooth-mdw, 2017-01-31:
@sandeepmistry where did that hex file come from? It doesn't look like one of mine based on the name.....

curioussavage, 2017-01-31:
@finneyj  My code isn't doing anything but subscribing to the characteristics for each button

curioussavage, 2017-01-31:
Also worth noting that there were less than a dozen button presses.

curioussavage, 2017-01-31:
If someone can point me to another firmware that would work I could try again tonight.

bluetooth-mdw, 2017-02-01:
Try the microbit Blue "no pairing required" hex file from Bitty Software: http://www.bittysoftware.com/downloads.html#microbit_blue

curioussavage, 2017-02-01:
@bluetooth-mdw  looks like that firmware displays a "C" char on the led 'display'. Thats 9 leds on constantly and about 20 minutes after connecting its still on.

bluetooth-mdw, 2017-02-01:
Sorry, forgot about that. Let me find you something else.....

bluetooth-mdw, 2017-02-01:
Try this one. Contains the following services:

```
    new MicroBitAccelerometerService(*uBit.ble, uBit.accelerometer);
    new MicroBitButtonService(*uBit.ble);
    new MicroBitIOPinService(*uBit.ble, uBit.io);
    new MicroBitLEDService(*uBit.ble, uBit.display);
    new MicroBitMagnetometerService(*uBit.ble, uBit.compass);
    new MicroBitTemperatureService(*uBit.ble, uBit.thermometer);
```

[microbit-ble-general-no-pair-p6.zip](https://github.com/sandeepmistry/node-bbc-microbit/files/743938/microbit-ble-general-no-pair-p6.zip)

The display is used but text scrolls off so it's only there briefly.

curioussavage, 2017-02-01:
thanks! will try right now

curioussavage, 2017-02-01:
Ok. Microbit is connected to my noble app and button presses are working! Now to go press the buttons for a day or two!

bluetooth-mdw, 2017-02-01:
Sounds like fun! :-)

fyi I've had a micro:bit plugged into a fresh battery since yesterday at 9am. No connection but it is advertising. Will see how it goes. It's suspected there's an issue with the hex file in this repo. Still waiting for @sandeepmistry to say where it came from as maybe he has the source code...? Or if it came from me and was renamed then maybe I have the source code....

finneyj, 2017-02-01:
Guys, for info - I just measured a micro:bit's power consumption running @bluetooth-mdw's HEX file. Under a BLE connected state, it's clocking in around 2.5mA @ 3v. That's a little high, but given that the HEX file is running all the sensor services, all the device drivers will be up (so it'll still be strobing the display, reading I2C from the sensors etc etc).

Even at that, you should be looking at 5 days or so of a CR2032 I would have thought. I have clocked a micro:bit consuming only 400uA though, when unnecessary sensors are disabled... so Maybe we should be also be looking at some HEX files (maybe generating through PXT @bluetooth-mdw?) that have only the driver you need...

bluetooth-mdw, 2017-02-01:
I'm afraid the micro:bit hex file creation service has been shut down ;-)

People wanting a hex file with only those Bluetooth services that they need, are hereby recommended to visit this lovely site and use the marvellous PXT tool: https://pxt.microbit.org/

sandeepmistry, 2017-02-02:
Firmware source can be found here: https://github.com/sandeepmistry/node-bbc-microbit/issues/6#issuecomment-255612167

It's based on the mbed (web) sample + has pairing disable + all services enabled and DFU removed ...

curioussavage, 2017-02-02:
Ok guys it was dead again this morning.

The microbit is connected to a device in an adjacent room with the door closed. Will that affect power usage?  How can I measure what it's drawing?

finneyj, 2017-02-03:
Folks,

I just profiled the microbit-dal rc7 firmware to see where the power goes. The results are quite interesting, and rather surprising! With all results in micro-amps, and listed in order of significance, here's where the power goes on a micro:bit:

[image]

In other words, we can probably get down to about 800uA (a third of what we currently have) by being a bit smarter about when we bring up the peripherals - especially the ones that aren't really used. I'll take that back as an action...

@curioussavage This still doesn't really explain why your seeing such poor results. It does sound like there's more going on there. To answer your question, you'd really need to put an ammeter between the battery and board to see how much power is being drawn...