sandeepmistry/node-bbc-microbit #15: MicroBitLightSensor

MatthijsKamstra, 2017-03-28:
This is not really an issue... but I am just curious...

I know that the microbit doesn't have a default light sensor, but it uses a trick to get that value: 
https://lancaster-university.github.io/microbit-docs/extras/light-sensing/

Is it possible with node.js?

smartyw, 2017-03-28:
It is if you're prepared to do some coding on the micro:bit itself using its C/C++ APIs. Per the page you linked to, you could read the light sensor periodically and each time you obtain a value, package it in a MicroBitEvent object with a ID of your choosing and the light sensor reading in the event' value part. Then have your node.js code subscribe to your chosen event ID and the "ANY" value wildcard and you should receive events in your node.js code, with the light sensor reading packaged in it.

sandeepmistry, 2017-04-10:
@smartyw thanks for answering!

@MatthijsKamstra the answer above be @smartyw looks good to me, so closing this.

closed 2017-04-10 by sandeepmistry