sandeepmistry/node-bbc-microbit #17: Combining node-bbc-microbit and rolling-spider

SebastienVermandel, 2017-09-26:
I would like to control a parrot drone using the sensors of a micro:bit.
I tried to combine the two libraries listed in the subject, but I am not able to get both BLE peripherals activated. I have a CSR dongle. Discovery seems to be ok. It gets complicated when trying to connect.

Help would be appreciated...

smartyw, 2017-09-27:
Is the drone a Bluetooth GAP Peripheral? If so, it's not possible to communicate directly with it from a microbit since it too is a GAP Peripheral. You would need a GAP Central device like a smartphone app in between the other two devices, acting like a hub. 

See http://bluetooth-mdw.blogspot.ch/2016/07/microbit-and-bluetooth-roles.html?m=0 for more on this and perhaps also http://www.bittysoftware.com/troubleshooting.html

sandeepmistry, 2017-10-02:
@bluetooth-mdw thanks for helping out!

@SebastienVermandel I would suggest you setup the connection to the micro:bit, and then call the rolling-spider API's:

```js
var BBCMicrobit = require('../index'); // or require('bbc-microbit')

var period = 160; // ms

console.log('Scanning for microbit');
BBCMicrobit.discover(function(microbit) {
  console.log('\tdiscovered microbit: id = %s, address = %s', microbit.id, microbit.address);

  microbit.on('disconnect', function() {
    console.log('\tmicrobit disconnected!');
    process.exit(0);
  });

 
  // add other micro:bit listeners ...

  console.log('connecting to microbit');
  microbit.connectAndSetUp(function() {
   // setup some micro:bit sensors ...

   // do some rolling-spider stuff ....
  });
 });
```

closed 2017-10-02 by sandeepmistry