sandeepmistry/node-bbc-microbit #18: Analog pins should be 10bits

jaafreitas, 2017-10-03:
According to the [documentation](https://www.microbit.co.uk/functions/analog-read-pin), analog pins should return a number between 0 and 1023 (included).

lib/io-pin-service.js is reading data as unsigned integer 8 bits

Steps to reproduce:
a) plug a potentiometer in P0
b) start example/pin-listener.js
c) change the potentiometer from min to max.
We will have 4 turns of 0 to 255 values.

It seems that the problem starts with the firmware, because we are receiving only 2 bytes.

Any workaround or suggestion would be great.

jaafreitas, 2017-10-03:
I read a bit of the [node-bbc-microbit-firmware](https://github.com/sandeepmistry/node-bbc-microbit-firmware) and the problem seems to be in the [microbit/microbit-dal/inc/bluetooth/MicroBitIOPinService.h/IOData struct](https://github.com/sandeepmistry/node-bbc-microbit-firmware/blob/a577c6319634d333e39240fcbe0d7fee08f9ac85/microbit/microbit-dal/inc/bluetooth/MicroBitIOPinService.h#L46).

```
struct IOData
{
    uint8_t     pin;
    uint8_t     value;
};
```

I have no ideia if changing the value to a uint16_t would be enough to solve this problem in the firmware.

jaafreitas, 2017-10-04:
According to the [official micro:bit documentation](https://lancaster-university.github.io/microbit-docs/ble/profile/#characteristics_4), the io pin service should truncate the analog values to 8bits.

I fixed it by dividing the getAnalogValue() by 4 on [onDataRead](https://github.com/sandeepmistry/node-bbc-microbit-firmware/blob/a577c6319634d333e39240fcbe0d7fee08f9ac85/microbit/microbit-dal/source/bluetooth/MicroBitIOPinService.cpp#L222) and [idleTick](https://github.com/sandeepmistry/node-bbc-microbit-firmware/blob/a577c6319634d333e39240fcbe0d7fee08f9ac85/microbit/microbit-dal/source/bluetooth/MicroBitIOPinService.cpp#L268)

This firmware is based on an old code and [things have changed a bit](https://github.com/lancaster-university/microbit-dal/blob/e0f8b005fe628552ab060a8f4cbbd1037bf3b8bb/source/bluetooth/MicroBitIOPinService.cpp#L247).

It seems that the node-bbc-microbit code is fully compatible with the [Microsoft MakeCode](https://makecode.microbit.org/_EfT8jeDcD8oW). I'm using it for now.

[image]

jaafreitas, 2017-10-04:
Just in case, if someone decided to fix this old firmware revision, the [code is available here](
https://os.mbed.com/users/jaafreitas/code/microbit-dal/rev/7c635e51760e)

smartyw, 2017-10-05:
I think this was already fixed in the DAL:

https://github.com/lancaster-university/microbit-dal/issues/289

jaafreitas, 2017-11-01:
Yes, it's already fixed in the DAL. Sadly not in the way I would expect it...

sandeepmistry, 2017-11-13:
@jaafreitas that documentation (from https://github.com/sandeepmistry/node-bbc-microbit/issues/18#issue-262318174) is not related to the firmware and BLE interface this library expects.

closed 2017-11-13 by sandeepmistry

jaafreitas, 2017-11-13:
I'm not sure why you closed this. You couldn't reproduce the problem in the description or you don't think it should be fixed?