sandeepmistry/node-bbc-microbit #20: Reading/Writing from UART

Lechucico, 2018-02-07:
I'm currently developing with a Micro:Bit. I need to read the accelerometer information and send over Bluetooth with a paired device when a huge change. I'm on a macbook, and I don't know how to read/write from UART. 

Does this node library works for read/write for the Micro:Bit? I would like to be able to read on the macbook the messages sent over Bluetooth with UART from the Micro:Bit.

Thanks!

bluetooth-mdw, 2018-02-07:
You don't need the UART service. The microbit profile gives you direct access to the accelerometer via the accelerometer service. If you only want accelerometer data when there is a "huge change" then I recommend using the microbit event characteristic to send the accelerometer value as a notification.

Take a look in the examples folder for help and see https://lancaster-university.github.io/microbit-docs/ble/profile/ for microbit profile information.

Lechucico, 2018-02-07:
My idea is the following. Since using Bluetooth consumes battery, I would like to accumulate information in the Micro: Bit and send it via UART.

If I use the accelerometer service, the macbook is the one that asks for the information (don't know how to do this) and I need to be making continuous requests to see what is happening in the Micro: Bit and that consumes energy.

bluetooth-mdw, 2018-02-07:
This is Bluetooth Low Energy so it's pretty power efficient. The mac does not need to poll the microbit. It will merely request 'notifications' once and then and wait for them to arrive. The microbit will only send a notification when it needs to. For your use case, monitor accelerometer values in microbit code and then send significant values as notifications relating to the microbit event characteristic. If you're not familiar with these terms, go to https://www.bluetooth.com/develop-with-bluetooth/developer-resources-tools/developer-kits/bluetooth-starter-kit and download the kit and read the basic theory document. 

Bitty Software have tutorials with code that show how to send microbit event characteristics:

http://bittysoftware.com/tutorials.html

The UART Bluetooth capability of the microbit is not UART, despite the name, so don't let that confuse you.

Lechucico, 2018-02-07:
So, the only way of sending accelerometer data from Micro:Bit to another device is over the accelerometer service?

There has to be a way of sending custom messages from Micro:Bit over Bluetooth no? 

I'll take a look at both links, thanks!

bluetooth-mdw, 2018-02-07:
No, not at all. "microbit event" is the name of a Bluetooth characteristic and is part of the Event Service. I'm recommending you use this. Have a look at the Bitty Software Data Logger code to see how microbit event is used to send analog pin values. You could use it in the same way but for your "substantial accelerometer" values, which you will sample in a loop on the microbit. 

I think you need to do some reading to get a clear idea of the basic Bluetooth LE concepts and the profile used by the microbit. The links I've shared will cover everything you need I think. 

Good luck :-)

Martin

Lechucico, 2018-02-08:
Hello @bluetooth-mdw. Thanks for all the links you provided to me, now I understand how BLE works. Now I see that there are services that has characteristics, and you can read those characteristics (or even write). 

Now I'll explain you my idea and I hope you can send me some tips:

The idea is to put the micro:bit on a patient on the hospital (on the arm). While he's not moving, don't send anything. I'll perform this checking accelerometer data or even when shake event happens. Once he starts moving, send to the paired device the accelerometer data. So, I have some questions:

1. With accelerometer service, my android application will get notifications everytime characteristic get updated (this is done with a period). This works with a period, so I'm no able to update the characteristic on demand, right?

2. With event service, I can get notifications when some event happens on micro:bit and also get the accelerometer data.

3. Using UART, I can get notifications when I programatically send some message (update the RX caracteristic).

So, it seems that the best aproach for sending accelerometer data on demand is UART, no?

bluetooth-mdw, 2018-02-08:
(2) happens only for event IDs that your Pi code is interested in. It must write to the Client Requirements characteristic and then subscribe to notifications on Microbit Event. So it's preyy much equivalent to (3) really.

sandeepmistry, 2018-02-10:
Using the UART for I/O is out of scope of the original micro:bit firmware source, since it has a service to bridge UART <-> BLE Characteristic.

Closing for now, but feel free to continue the discussion here.

closed 2018-02-10 by sandeepmistry

bluetooth-mdw, 2018-02-10:
@lechucio like I said, use the event service. microbit programming is event based so it's very easy to create an event object and it be transmitted as a Bluetooth notification.

Lechucico, 2018-02-19:
Hello @bluetooth-mdw . I've been developing an Android application for connecting to the Micro:Bit device. I'm just obtaining 1 service and 0 characteristics from the Micro:Bit. Can you take a look please? https://stackoverflow.com/questions/48871685/obtain-notifications-from-bluetooth-gatt-characteristic